react-native-sensors/react-native-sensors #294: Help Please - Gyroscope or Accelerometer?

damialajogun, 2019-12-30:
Hi everyone,

I am working on a charades-inspired game (just like Heads Up!) where the user tilts their phone down if they guess a word correctly or up to pass. I'm confused about which sensor will work best for this logic. I assume Gyroscope, since that measures rotational changes and I am looking to pull the next card only if the phone tilts up or down, however, when I added this into my code, the cards change sporadically. 
Then, I tried Accelerometer but when I tilt my phone down, pretty much all the cards run up in a matter of seconds (I'm guessing because the motion is in the direction of gravity?). It could also be the z-values I have are not correct. 
Any suggestions/help you can provide is greatly appreciated. Thank you!

[image]
[image]

DanielMSchmidt, 2020-01-02:
The sensor gives you just the raw data, so just how fast you are tilting, not how much tilt you currently have. If you want to have a "look down" computation you need the values added up over time to see at which state in relative to the start you are. I personally get x,y,z confused all the time, so I would suggest doing the adding for all values separately and logging it to see which one changes the most.

Let me know if this works for you. I would really appreciate it if you could as soon as you found a solution write a tutorial into our docs :) I am sure it will help a lot of people!

damialajogun, 2020-01-02:
Hi @DanielMSchmidt , thanks for responding! I think you might be onto something here, but I'd like to make sure I understand what you're suggesting. 

> If you want to have a "look down" computation you need the values added up over time to see at which state in relative to the start you are.

Can you please explain what you're thinking here a bit more? If the values are how fast the phone is tilting (so rates), how does the difference in rates tell me which state I am in? Wouldn't the result be a rate as well? What I'm trying to achieve with the logic is that the card displayed on the screen changes after the phone is tilted down, so would using the difference in values be inconsistent depending on how quickly or slowly a person tilts their phone? I could be misunderstanding.

> Let me know if this works for you. I would really appreciate it if you could as soon as you found a solution write a tutorial into our docs :) I am sure it will help a lot of people!

Absolutely! I'd definitely share a solution if I find one :)

closed 2020-02-09 by stale[bot]