react-native-sensors/react-native-sensors #295: Cannot use Accelerometer in functional component

Rasmustof, 2020-01-03:
**Describe the bug**
when attempting to write an app that just displays data from the accelerometer of my phone. the setUpdateInternvalForType function appears to trigger a "TypeError" as shown in the provided screenshot. 

**To Reproduce**
Steps to reproduce the behavior:

1.  Copy example code into a project
2. Run app
3. See error

**Expected behavior**
the app should display a text object with a number from the accelerometer of the phone

**Desktop (please complete the following information):**

* React Native Version: 0.61.5
* Device: Samsung s9+ 
* Version: 5.3.3

**Additional context**
I have added the file I made into this. and i also get a diffrent error if i just comment out the "setUpdateIntervalForType" call. about it not know what React is ?!? seems odd. 

CODE: https://pastebin.com/JwvgtXsE

Rasmustof, 2020-01-03:
[image]
sorry, i forgot to add the screenshot

DanielMSchmidt, 2020-01-03:
I see, we should improve the validation / error messages. If you have time I think this would be a great first PR :) 

So the problem is that it has to be `SensorType.accelerometer` (https://github.com/react-native-sensors/react-native-sensors/blob/fa1cb601a795247ca0eb53d24c697c899297da33/examples/KeepTheBallGame/App.js#L11)

Because you have the wrong type there you pass an undefined, which can't do toLowerCase.

Best would be if we could add checks at these places, so the error message is not that cryptic: https://github.com/react-native-sensors/react-native-sensors/blob/ab93f61a69294002bea58ec0b4dc4b81ee88b321/src/rnsensors.js#L27

Rasmustof, 2020-01-03:
oh yea, that seems right. would be more helpful with a proper error message yes.
as with the PR, im currently making a project for my exam, so i wont be able to work on that now, sadly

Rasmustof, 2020-01-03:
but thanks for the quick response

closed 2020-02-09 by stale[bot]