react-native-sensors/react-native-sensors #298: Log spamming

eventengineering, 2020-01-23:
**Describe the bug**
An unappreciated number of log entries are being output

**To Reproduce**
Subscribe to the accelerometor with a function that can do absolutely nothing, and watch as your log becomes unusable

**Expected behavior**
Respectful amounts of logging: none happening on each read, maybe some on enabling and some on disabling.

**Desktop (please complete the following information):**

* React Native Version: 0.61.1
* Device: Any iOS device. Replicated on iPhone XS, iPhone 8+, iPad 2018
* Version: 5.3.3 (and I can see the lines on your main branch)

**Additional context**
https://github.com/react-native-sensors/react-native-sensors/blob/5107de17147458ba76784350732f2a67ec0e4383/ios/Accelerometer.m#L101

DanielMSchmidt, 2020-01-23:
Feel free to add a PR to make this conditional 👍

eventengineering, 2020-02-05:
I don't think this even needs to be conditional: if someone wants to log it out, they can in the listener, there's no need for a package to log so verbosely except when developing.

DanielMSchmidt, 2020-02-05:
I disagree. If for whatever reason the message bus between native and js is clogged one can without a log not see if the emitter still emitted events

eventengineering, 2020-02-05:
I see your point, but I have no idea about how to go about adding a conditional flag or how you would want the conditional flag to even work. Unfortunately, the verbose logging at present makes this package unusable.

What should the default option be? My recommendation is for it to be off for normal operation as every log in RN slows the app down and ordinarily in a live environment you won't need the logs.

Should this be an optional parameter on the `setUpdateIntervalForType` function, or a shared flag for all the sensors set through another function?

DanielMSchmidt, 2020-02-05:
I think something similar to `setUpdateIntervalForType`, so maybe `setLogLevelForType` would be a great API, defaulting to not logging anything.

SnaiNeR, 2020-02-23:
there is already rejected PR [https://github.com/react-native-sensors/react-native-sensors/pull/246](https://github.com/react-native-sensors/react-native-sensors/pull/246)
i will reply here too, 
> this logs are spamming so much in logs window, if you work with them directly - you can enable them directly in ios native code, or just use console log in js code

DanielMSchmidt, 2020-02-24:
As I said, I'd like native logs to be configurable

eventengineering, 2020-03-03:
You've requested "log level", so I'm aiming for the following:

0: no logging
1: normal level of logging, nothing too verbose
2: log everything

Specifically, the log line that happens with every reading is restricted to the second level.

These are easily changeable in the future.

DanielMSchmidt, 2020-03-03:
Sounds great 👍

eventengineering, 2020-03-09:
Problem being "no environment variables set" bug with Travis-ci.

SnaiNeR, 2020-03-15:
good job @eventengineering

closed 2020-04-14 by eventengineering