react-native-sensors/react-native-sensors #302: Accelerometer subscribe doesn't give right data

borisfuture128, 2020-03-05:
I tried to get sensor data using Accelerometer.
But 

                                  accelerometer.subscribe(
					({ x, y, z }) => {
						this.setState({
							x: x,
							y: y,
							z: z
						})
					},
					(err) => {
						this.setState({ error: err })
					},
				);

this is code.
it gives dynamic values, x, y, z.
But they are different from log of native ios.
As a result, can't use this sensor.
I guess this is bug or I made something wrong somewhere.
Please help me.

Best regards

borisfuture128, 2020-03-05:
my react native version is 0.46.1

borisfuture128, 2020-03-05:
I tried demo gyroscopeimage as well

borisfuture128, 2020-03-05:
but it didn't work:(

DanielMSchmidt, 2020-03-05:
Can you explain how the values are different from the log? How did you get the native log?

borisfuture128, 2020-03-10:
by Xcode

closed 2020-04-17 by stale[bot]