react-native-sensors/react-native-sensors #311: Is there a Heartbeat sensor?

yesIamFaded, 2020-03-25:
Hi guys,
I have a question or request if its not implemented yet.
On my Samsung S9 next to the camera I have a Fingerscanner that can tell me my heartrate for example.

I want to build a small FitnessApp and I would like to use this heartbeat detection.
Can I somehow get access to this sensor by using this package? OR maybe anyone knows how to access it? 
Ty - Stay Safe 😷

DanielMSchmidt, 2020-03-25:
I think there is:

- [iOS](https://developer.apple.com/documentation/healthkit/hkheartbeatseriesquery)
- [Android](https://developer.android.com/reference/android/hardware/Sensor#STRING_TYPE_HEART_RATE)

Would you be interested in adding it to React Native Sensors? Would be happy to review a PR :)

yesIamFaded, 2020-03-26:
Honestly I wouldnt even know how to do it.. :D 

So just to be clear the module currently doesnt provide this feature and I would have to tackle some native java code right?

yesIamFaded, 2020-03-30:
hello? 😟

DanielMSchmidt, 2020-03-30:
Yes, the current module does not provide this feature and if you want to enhance `react-native-sensors` you would need to write Objective-C and Java code. If you just want to solve this in your codebase java code should be enough

yesIamFaded, 2020-03-30:
@DanielMSchmidt alright ty - do you know if there is a guide or something to this?

DanielMSchmidt, 2020-03-30:
I linked two [above](https://github.com/react-native-sensors/react-native-sensors/issues/311#issuecomment-603924893)

yesIamFaded, 2020-03-30:
Aight thanks again, I am going to close this then.

closed 2020-03-30 by yesIamFaded