react-native-sensors/react-native-sensors #317: Jest doesn't render the app

apgsn, 2020-05-08:
Hi,  
I'm working on a React Native app (v. `0.62.2`, no Expo) which will use the v. `5.4.1` of React Native Sensors.  
To test things out, in `App.js` I added the [usage sample code](https://react-native-sensors.github.io/docs/Usage.html) to a `useEffect` hook which only runs on mount/unmount, and after a small fix (using `SensorTypes.accelerometer` instead of `SensorTypes.Accelerometer`) the app is now working fine, both on my Android emulator and physical device. I didn't test it on iOS yet.  
Today I tried using Jest, which is shipped by default with RN, but the initial test ('renders correctly') doesn't seem to work. The error I get is:
```
Test suite failed to run

    Native modules for sensors not available. Did react-native link run successfully?

      at Object.<anonymous> (node_modules/react-native-sensors/src/rnsensors.js:10:9)
      at Object.<anonymous> (node_modules/react-native-sensors/src/sensors.js:4:1)
```
which refers to the code in the module `rnsensors.js`:
```javascript
if (!GyroNative && !AccNative && !MagnNative && !BarNative) {
  throw new Error(
    "Native modules for sensors not available. Did react-native link run successfully?"
  );
}
```
And even if i force this condition to `false`, I have other type of errors, such as:
```
  console.error node_modules/react-test-renderer/cjs/react-test-renderer.development.js:11944
    The above error occurred in the <App> component:
        in App
    
    Consider adding an error boundary to your tree to customize error handling behavior.
    Visit https://fb.me/react-error-boundaries to learn more about error boundaries.

TypeError: Cannot read property 'setUpdateInterval' of undefined
```

I'm new to Jest so nevermind testing this part of the code, I just want to have it at least up and running to try it out. How do I fix this?

DanielMSchmidt, 2020-05-20:
I don't think we support this currently, I would advice you to [mock](https://jestjs.io/docs/en/manual-mocks) the API for unit tests. I hope this helps, if not feel free to reopen :)

closed 2020-05-20 by DanielMSchmidt

CharlesMangwa, 2021-02-12:
For anyone would ended up in this issue and just wants a fast copy/paste, I quickly came up with this based on [the TS interface](https://github.com/react-native-sensors/react-native-sensors/blob/390b56150f7abe6055943f5b2758dcf5293ae8e0/index.d.ts): 
```js
jest.mock('react-native-sensors', () => ({
  accelerometer: {
    subscribe: () => ({
      unsubscribe: jest.fn(),
    }),
  },
  gyroscope: {
    subscribe: () => ({
      unsubscribe: jest.fn(),
    }),
  },
  magnetometer: {
    subscribe: () => ({
      unsubscribe: jest.fn(),
    }),
  },
  barometer: {
    subscribe: () => ({
      unsubscribe: jest.fn(),
    }),
  },
  SensorTypes: {
    accelerometer: 'accelerometer',
    gyroscope: 'gyroscope',
    magnetometer: 'magnetometer',
    barometer: 'barometer',
  },
  setUpdateIntervalForType: jest.fn(),
}))
```

CharlesMangwa, 2021-02-12:
For anyone who ended up in this issue and just wants a fast copy/paste, I quickly came up with this based on [the TS interface](https://github.com/react-native-sensors/react-native-sensors/blob/390b56150f7abe6055943f5b2758dcf5293ae8e0/index.d.ts): 
```js
jest.mock('react-native-sensors', () => ({
  accelerometer: {
    subscribe: () => ({
      unsubscribe: jest.fn(),
    }),
  },
  gyroscope: {
    subscribe: () => ({
      unsubscribe: jest.fn(),
    }),
  },
  magnetometer: {
    subscribe: () => ({
      unsubscribe: jest.fn(),
    }),
  },
  barometer: {
    subscribe: () => ({
      unsubscribe: jest.fn(),
    }),
  },
  SensorTypes: {
    accelerometer: 'accelerometer',
    gyroscope: 'gyroscope',
    magnetometer: 'magnetometer',
    barometer: 'barometer',
  },
  setUpdateIntervalForType: jest.fn(),
}))
```

DanielMSchmidt, 2021-02-18:
That's cool @CharlesMangwa can you do a PR to the readme so it's easier to find for newcomers?

CharlesMangwa, 2021-02-19:
Sure thing, can do! Do you think it should be added to the README or somewhere on the website? Seems like all the library docs are there. Just lemme know what would be preferable in your opinion.