react-native-sensors/react-native-sensors #319: Light Sensor

rozekfr, 2020-05-15:
**What is the purpose of this sensor?**
I would like to use a sensor to detect lighting conditions. Based on these conditions, I would like to change the graphics of the application from light to dark mode something like Google Maps for navigation.

**Is it supported on both iOS and Android?**
I am using react-native-sensor-manager for detecting ambient light on Android. You mentioned this library like inspiration at documentation.
As for iOS it is very strange. IOKit supports this sensor, but Apple will reject the application, because it is forbidden to manipulate with brightness, but it is not what I would like to do. More information is here. [here](https://forums.developer.apple.com/thread/36512).

Maybe you (we) can add support for Android and in the meantime, we can find out some information about Apple's ambient light sensor.

iOS workaround (an idea). Most of modern devices are setting brightness automatically by the ambient light sensor. So maybe there is an opportunity to use the brightness value to determine ambient light.  I think that this workaround should pass Apple review, but it will not be ideal.

`let currentBrightness = UIScreen.mainScreen().brightness`

**Is there a way to test this sensor without a real device?**
Unsure

**Would you be willing to respond to issues / PRs for this sensor?**
Unsure

DanielMSchmidt, 2020-05-15:
Hey, yeah that sounds like a good thing to add.

- There is an android implementation: https://developer.android.com/reference/android/hardware/Sensor#TYPE_LIGHT
- We have testability on the android side: https://stackoverflow.com/questions/40143578/is-the-current-provided-android-emulator-able-to-simulate-the-sensors
- It's not testable in iOS as far as I see: https://www.browserstack.com/test-on-ios-simulator

I'd love to review a PR for this 👍 
Here are some common traps we ran into on the last few sensors:

- The unit might be different Android vs iOS, we need to normalize it, ideally on the native side
- We need to test this both on iOS and Android, wouldn't want to ship with a runtime error when the sensor is used
- We need to ensure the docs reflect the new sensor and provide the configuration options

organom, 2020-06-28:
this is still relevant

DanielMSchmidt, 2020-08-16:
Still relevant, right @organom?

Do you see a chance to devote any time into a PR for this?

organom, 2020-08-20:
@DanielMSchmidt I looked briefly into it and hit the Apple API that could bring apps to be rejected on the app store, need to invest some more time researching around that API, hopefully we can read the values (there or somewhere else). In the android side it seems trivial. Will try to invest time into it in a near future :)

DanielMSchmidt, 2020-08-20:
Awesome @organom 🎉 I mean we would not want our users be rejected just for having the code in, but one would old be if it's used, right?

spamingdnl, 2020-08-24:
I've been looking arround and I couldn't find any npm package implementing the Light Sensor on Android on React Native 0.60+.
This feature would be very usefull for me and the React Native community I think. I tried to write it myself on android with NativeModule but I'm a beginner and understand  nothing about writing Native Module

organom, 2020-10-04:
Still didn't had time to look into it :( hopefully will manage to get some time soon

organom, 2020-11-08:
unstale :) will look into this

Maximvdw, 2021-03-05:
would also be very much interested in this +1

anutting, 2021-03-11:
Also interested in using this if it ends up being implemented

ZweiEuro, 2022-08-03:
@organom Are you still working on this or have you found other relevant information ?

organom, 2022-08-03:
@ZweiEuro unfortunately my free time in the last year has been close to 0, and to my best estimation I'll not be able to get back to this anytime soon. If someone wants to pick the task, please feel free.

zakwilson, 2023-08-06:
It appears Apple no longer forbids using the light sensor on iOS (I am not an expert on Apple policies though).

https://developer.apple.com/documentation/sensorkit/srambientlightsample

srivastava9, 2024-09-27:
Has anyone picked this up? I want to auto switch flash on/off depending on surrounding light , using vision camera

DanielMSchmidt, 2024-09-27:
There have been a lot of PRs on this repo for this feature, but since only one platform supports it and the central idea of this library is universal support we could never merge them. I bet you can get what you need out of these PRs and make a separate library out of it. I'd be happy to link to it from our readme as well 👍