react-native-sensors/react-native-sensors #327: Bridge is not set

Serzhs, 2020-08-05:
When I reload app I get this error:
`*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Error when sending event: Gyroscope with body: {
    timestamp = 1596624793328;
    x = "0.2605170607566833";
    y = "0.1583694070577621";
    z = "-0.1048433259129524";
}. Bridge is not set. This is probably because you've explicitly synthesized the bridge in Gyroscope, even though it's inherited from RCTEventEmitter.'
*** First throw call stack:
(0x199224300 0x198f38c1c 0x1991211f8 0x1995b75c4 0x104d2d7b4 0x104cc2508 0x1995f7cfc 0x1994f9c3c 0x1995fa03c 0x1994f9914 0x1995faa68 0x1995fa528 0x10755a5dc 0x10754b18c 0x107559d14 0x19919f6b0 0x19919a2c8 0x1991998f4 0x1a35b0604 0x19d36d358 0x10471de84 0x1990152dc)`

"react-native": "0.62.2",
"react-native-sensors": "^6.1.0",

My code:
`setUpdateIntervalForType(SensorTypes.gyroscope, 100);
    useEffect(() => {
        gyroscope.subscribe(({ x, y }) => {
            const wholeX = Math.floor(x);
            const wholeY = Math.floor(y);
            if(wholeX !== gyroscopePosition.x || wholeY !== gyroscopePosition.y) {
                setGyroscopePosition({
                    x: wholeX,
                    y: wholeY,
                });
            }
        });
    }, []);`

reinismu, 2020-08-05:
It could be something related to https://github.com/react-native-sensors/react-native-sensors/pull/270

closed 2020-09-12 by stale[bot]

royderksroute42, 2020-12-11:
Hi,

I've also encountered this problem.

"react-native": "0.64.0-rc.1",
"react-native-sensors@^7.0.2 (7.02 in lockfile)"

All .m files seem to explicitly synthesize the bridge, and they aren't allowed to do this anymore apparently, as it's inherited.