react-native-sensors/react-native-sensors #339: Help please to do a subscription.unsubscribe()

emurigar, 2020-09-04:
Hi,

I'm working on the react-native versions 0.63.1 (no Expo) which I use the 7.0.0 of React Native Sensors.

 I'm try to do a subscription when the user enter in a screen and make a unsubscribe when the user leave the screen and go to another one, but I don't know how can I stop the accelerometer. I tried several ways to do it, but I didnt get it, my las code is the following

```
let subscription;
...
function startSensor(status){
    if(status) {
      subscription = (accelerometer.subscribe(({ x, y, z, timestamp }) =>
      (setAccelerometerX(x), setAccelerometerY(y))
      ));
    } else {
      subscription.unsubscribe();
    }
  }
```

DanielMSchmidt, 2020-09-30:
Sorry for the late response, what do you mean with it is not stopped?

emurigar, 2020-10-01:
Hi @DanielMSchmidt  thanks for answering. My problem is I don't know how to make the unsubscribe if I write the `subscription.unsubscribe()`in different method that I initialize the `subscription`.

I use subscription method when I enter a screen and I would like to use unsubscribe method when I leave it

Thanks!

DanielMSchmidt, 2020-10-01:
Oh now I get it, thank you 👍 I think this should work, made a small improvement to guard against subscriptions not being initialized yet.

```js
let subscription;
...
function startSensor(status){
    if(status) {
      subscription = (accelerometer.subscribe(({ x, y, z, timestamp }) =>
      (setAccelerometerX(x), setAccelerometerY(y))
      ));
    } else {
      if (subscription) {
        subscription.unsubscribe();
      }
    }
  }
```

emurigar, 2020-10-02:
Thanks @DanielMSchmidt I just tried it and it was resolved!!

Regards!!

closed 2020-10-02 by emurigar