react-native-sensors/react-native-sensors #344: Supporting additional sensors like RotationRate, Attitude and Device Orientation

o16-waqas, 2020-09-24:
**Is your feature request related to a problem? Please describe.**
- Is it possible to support the additional sensors values like **RotationRate**, **Attitude**, **Device Orientation**?  

**Describe the solution you'd like**
- Like the currently supported sensors Gyroscope, Accelerometer, Magnetometer, Barometer, Above mentioned sensors can also be supported

**Describe alternatives you've considered**
Expo sensors (https://docs.expo.io/versions/latest/sdk/sensors/) provides these requested sensors values along with the sensors supported by react-native-sensors. I've used them using uni_modules but they still missing the timestamp of the event. 

**Additional context**
Described above.

DanielMSchmidt, 2020-09-24:
Hey @o16-waqas, love it!
Please keep in mind that one of the strict rules of this project is that each sensor need to be supported for both iOS and Android. As long as your PR contains both implementations I am more then happy to review it and get it in 💪

organom, 2020-11-08:
unstale

closed 2022-01-09 by stale[bot]