react-native-sensors/react-native-sensors #346: react-native-sensors Error: Native modules for sensors not available. Did react-native link run successfully?

bobdelgreco, 2020-10-20:
https://stackoverflow.com/questions/64437070/react-native-sensors-error-native-modules-for-sensors-not-available-did-react

Thanks for any help, getting error message "Error: Native modules for sensors not available. Did react-native link run successfully?" when running npx react-native start

npx react-native run-android gives this error: > Task :app:processDebugManifest FAILED right after installing react-native-sensors.

Install procedure:

npx react-native init proj8

cd proj8

npm install react-native-sensors --save

npx react-native link react-native-sensors

The last command reported successful but when I went through the manual Android procedure some parts were not done as follows:

import com.sensors.RNSensorsPackage;
was in android/app/src/main/java/[...]/MainApplication.java

new RNSensorsPackage() was not there, I added as follows:

@Override
       protected List<ReactPackage> getPackages() {
       @SuppressWarnings("UnnecessaryLocalVariable")
       List<ReactPackage> packages = new PackageList(this).getPackages();
       // Packages that cannot be autolinked yet can be added manually here, for example:
       // packages.add(new MyReactNativePackage());
       packages.add(new RNSensorsPackage());
       return packages;
}
include ':react-native-sensors'
project(':react-native-sensors').projectDir = new File(rootProject.projectDir,  '../node_modules/react-native-sensors/android')
were both there in android/settings.gradle

dependencies {
    implementation project(':react-native-sensors')
    implementation fileTree(dir: "libs", include: ["*.jar"])
    //noinspection GradleDynamicVersion
    implementation "com.facebook.react:react-native:+"  // From node_modules
I had to add implementation project(':react-native-sensors')

I don't think my code matters yet but App.js:

import React, {useState, useEffect, useRef} from 'react';
import { StyleSheet,ScrollView,SafeAreaView,Platform, View,Text } from 'react-native';

export default function App() {
    var Rndr = 
        <Text>asdf</Text>
    return Rndr
}
package.json:

{
  "name": "proj8",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "react": "16.13.1",
    "react-native": "0.63.3",
    "react-native-sensors": "^7.0.1"
  },
  "devDependencies": {
    "@babel/core": "^7.8.4",
    "@babel/runtime": "^7.8.4",
    "@react-native-community/eslint-config": "^1.1.0",
    "babel-jest": "^25.1.0",
    "eslint": "^6.5.1",
    "jest": "^25.1.0",
    "metro-react-native-babel-preset": "^0.59.0",
    "react-test-renderer": "16.13.1"
  },
  "jest": {
    "preset": "react-native"
  }
}

ghirihari, 2020-11-20:
I encountered the same problem. This occurs because react-native-sensors has a minsdk version is 17 and react-native's default minsdk version is 16 Its quite easy to solve. 

 1. Just go to **android/build.gradle** 
 2. Change **minSdkVersion** to anything above 17 

Before

     ext {
        buildToolsVersion = "29.0.2"
        minSdkVersion = 16
        compileSdkVersion = 29
        targetSdkVersion = 29
    }

After

     ext {
        buildToolsVersion = "29.0.2"
        minSdkVersion = 19
        compileSdkVersion = 29
        targetSdkVersion = 29
    }

closed 2022-01-09 by stale[bot]

carltian, 2022-03-07:
same issue, how do u solve it? i try to change minSdkVersion, but it doesn't work.

mehandi-bytes, 2023-10-23:
> same issue, how do u solve it? i try to change minSdkVersion, but it doesn't work.

yaa i have same issue but doesn't resolve @carltian do you find any solution