react-native-sensors/react-native-sensors #353: Different values between iOS and Android?

kirkryan, 2021-02-05:
**Describe the bug**
When displaying the accelerometer values in the demo code, iOS reads around -10 when placed static on a desk, and android reads 0 when placed the same.

**To Reproduce**
Steps to reproduce the behavior:

`const subscription = accelerometer
  //.pipe(map(({ x, y, z }) => x + y + z), filter(speed => speed > 20))
  .subscribe(
    speed => console.log(`You moved your phone with ${speed}`),
    error => {
      console.log("The sensor is not available");
    }
  );`

1.  Run on iOS and Android
2. Values are different when static

**Expected behaviour**
I assumed that as a universal library there would be some sort of normalisation to ensure that the readings would be universal between the platforms )i.e. uniformly use values with gravity or without gravity. In this instance, it seems like iOS is including gravity, and android is not.

* React Native Version: [0.63.4]
* Device [iPhone 12 Pro and Pixel 4a]
* Version [iOS14.4 and Android 11]

**Additional context**
Add any other context about the problem here.

DanielMSchmidt, 2021-02-08:
I wonder if this could be related to the issue mentioned here: https://github.com/react-native-sensors/react-native-sensors/issues/26#issuecomment-763784947

kirkryan, 2021-02-08:
Ah! Yes, that's it indeed. I added the ternary multiplication at the end to do the conversation - is it worth adding this as a footnote in the instructions to prevent others from getting confused?

DanielMSchmidt, 2021-02-08:
I think it would be best to fix this Inside the Library, would you have the
time to do a PR?

kirkryan, 2021-02-08:
Sure!

Kirk Ryan

Co-founder & Technical Director

BUSBY - NEVER ALONE

kirkryan, 2021-02-09:
Hi, I just realised that the code is not javascript, I am not familiar with swift or java so will struggle with this request for a PR.

DanielMSchmidt, 2021-02-09:
You could solve it in JS if you want, should be doable via a map on the
observable switching the multiplication. Not an ideal solution, but as long
as it's inside this library that should be good enough

kirkryan, 2021-02-18:
Ah- this at first appearance seemed to be the same issue, however, it's actually iOS includes gravity (-9.81m/s) and android does not.