react-native-sensors/react-native-sensors #376: Calibrated Sensor Data

amirhmk, 2021-03-23:
**Is your feature request related to a problem? Please describe.**
I have installed and started using this module for a product that I'm working on, and it works seamlessly and was quite easy to setup. However, it seems that the sensor data are raw data that are coming from the sensors, and it's not performing the calibration that the native module offers (i.e. excluding gravity). In iOS this is done by [CMotionManager](https://developer.apple.com/documentation/coremotion/getting_processed_device-motion_data) and [Linear Acceleration](https://developer.android.com/reference/android/hardware/Sensor#TYPE_LINEAR_ACCELERATION) in Android. I think this is available in both Android and iOS, and should follow the philosophy of the repository. 

**Describe the solution you'd like**
It would be great to add functionality to the module so the data returned by the sensors is after the post processing rather than the raw data. Use the `deviceMotion` rather than the `raw acceleration`. 

**Describe alternatives you've considered**
Creating a native module

DanielMSchmidt, 2021-03-24:
Seems to make sense, it seems like it would make sense to expose both types of data though, there might be folks that are interested in the raw data. Would you have the time to do a PR?

amirhmk, 2021-03-24:
> Seems to make sense, it seems like it would make sense to expose both types of data though, there might be folks that are interested in the raw data. Would you have the time to do a PR?

Yep for sure, I think it has to be a choice. 

Sure, will try to take a stab at this this week 😄

gabrieltorresgamez, 2023-05-23:
> > Seems to make sense, it seems like it would make sense to expose both types of data though, there might be folks that are interested in the raw data. Would you have the time to do a PR?
> 
> Yep for sure, I think it has to be a choice.
> 
> Sure, will try to take a stab at this this week 😄

Hi @amirhmk 

I just wanted to ask if you were successful in implementing this feature or used an alternative.