react-native-sensors/react-native-sensors #381: react-native-orientation-locker OrientationModule Conflicting with react-native-sensors

the-one-jay-301, 2021-04-02:
**Describe the bug**
[image]

**To Reproduce**
Steps to reproduce the behavior:

1.  install react-native-orientation-locker & react-native-sensors
2.  When build in development throwing error as per screen shot above.

**Expected behavior**
Orientation module name should be renamed or should implement canOverrideExistingModule. It should run with both packages.

**Desktop (please complete the following information):**

- React Native Version: 0.63.4
- Device Android, Android simulator
- Version latest

stanlygamer, 2021-04-03:
Same error ;

I am using 

React Native Version: 0.63.4
Device Android, samsung with android 11

poorvajappja, 2021-04-03:
Same issue

PradeepSimba, 2021-04-03:
Same issue

the-one-jay-301, 2021-04-05:
@DanielMSchmidt Can you provide quick fix for this issue ?

ricardoballaro, 2021-04-11:
Hey guys, while we don't have a fix. I worked around the problem using the previous version.
Work for me.
npm i react-native-sensors@7.0.2

stanlygamer, 2021-04-13:
> Hey guys, while we don't have a fix. I worked around the problem using the previous version.
> Work for me.
> npm i react-native-sensors@7.0.2

thanks , it works.

polymorpher, 2021-04-23:
Is there any other way to fix it, without forking the whole repository and changing the file names?

https://github.com/react-native-sensors/react-native-sensors/compare/v7.0.2...v7.2.0

DanielMSchmidt, 2021-04-23:
Hey folks, I have a PR with the fix, can you please test the new pre-release: https://github.com/react-native-sensors/react-native-sensors/pull/383 and approve if its working as expected

MichHighlander, 2021-04-28:
Just so you know, android has the same problems

closed 2021-06-12 by DanielMSchmidt

bqrichards, 2021-06-14:
@DanielMSchmidt Although #383 fixed a build error with Xcode / iOS, this module error still appears on Android. It looks like the module named "Orientation" is taken by org.wonday.orientation.OrientationModule as well as com.sensors.RNSensor (when sensor of name "Orientation" is made). 

I would love to make a PR on react-native-orientation-locker because it looks like the amount of changes needed would be smaller, but I'm unsure if they're as active as this repo as been. Hm...

DanielMSchmidt, 2021-07-23:
:tada: This issue has been resolved in version 7.3.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-sensors/v/7.3.0)
- [GitHub release](https://github.com/react-native-sensors/react-native-sensors/releases/tag/v7.3.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 7e50b2f0 (2021-04-23): fix: prefix native module files

Closes #381
PR #383 (2021-06-12): Prefix native module files
Closes #381

Per Release has been made, please test: https://github.com/react-native-sensors/react-native-sensors/releases/tag/v7.2.1-rc.1
commit 7c75788d (2021-06-12): fix: prefix native module files

Closes #381