react-native-sensors/react-native-sensors #391: TYPE_LINEAR_ACCELERATION Exclude gravity from acceleration

leamachado7, 2021-06-07:
**What is the purpose of this sensor?**
Because some times you want to exclude gravity from calculations, it would be nice to have it there.

**Is it supported on both iOS and Android?**
Yes

**Is there a way to test this sensor without a real device?**
Yes

**Would you be willing to respond to issues / PRs for this sensor?**
Yes

pireez, 2021-07-16:
What this package represents right now is ( Gravity  + device acceleration ) = acceleration 
this week I will pull a request to add two more types
acceleration , gravity excluded = linearAcceleration
and "gravity" stand alone
1-acceleration
2-linearAcceleration
3-gravity

for android is ready and I'm working for ios to make it work so the package will be promising for both!

stay tuned ;)

DanielMSchmidt, 2021-07-17:
Sounds great, this is the most requested feature! Thanks a ton!

Ehsan0Hejazi, 2021-07-17:
@DanielMSchmidt still working on it
some phones does not support linearAcceleration natively but gravity is supported on all of them
so I will only add gravity to package so users can easily get 
linearAcceleration = acceleration - gravity

Ehsan0Hejazi, 2021-07-18:
gravity property added for both ios and android to the package and I'm waiting for you @DanielMSchmidt  to merge my pull request

you can even update the API doc
and add the gravity to it 
It's just like accelerometer outputs

DanielMSchmidt, 2021-07-24:
This seems to be solved now :)

closed 2021-07-24 by DanielMSchmidt

Tim4497, 2022-02-28:
Hey, I have a maybe more general question, but how do you subtract the gravity from the acceleration synchronous?

Im pulling both once per second but it seems that gravity starts a bit later than Accelerator so when I calculate the linear Acceleration it's slightly off

Ehsan0Hejazi, 2022-02-28:
Hey, nice question unfortunately site document is not updated yet and I will make an update for it,
but document inside the source code in updated by me:

here's the new gravity API added to it:
[https://github.com/react-native-sensors/react-native-sensors/blob/master/docs/API.md#gravity-observablex-number-y-number-z-number-timestamp-string](url)

and here I have explained exactly how it can be excluded by RxJS synchronous
[https://github.com/react-native-sensors/react-native-sensors/blob/master/docs/Usage.md#raw-device-acceleration](url)

if there is more question about it let me know ;)

Ehsan0Hejazi, 2022-02-28:
> Hey, I have a maybe more general question, but how do you subtract the gravity from the acceleration synchronous?
> 
> Im pulling both once per second but it seems that gravity starts a bit later than Accelerator so when I calculate the linear Acceleration it's slightly off

Hey, nice question unfortunately site document is not updated yet and I will make an update for it,
but document inside the source code in updated by me:

here's the new gravity API added to it:
[https://github.com/react-native-sensors/react-native-sensors/blob/master/docs/API.md#gravity-observablex-number-y-number-z-number-timestamp-string](https://github.com/react-native-sensors/react-native-sensors/issues/url)

and here I have explained exactly how it can be excluded by RxJS synchronous
[https://github.com/react-native-sensors/react-native-sensors/blob/master/docs/Usage.md#raw-device-acceleration](https://github.com/react-native-sensors/react-native-sensors/issues/url)

if there is more question about it let me know ;)

Ehsan0Hejazi, 2022-02-28:
> Hey, I have a maybe more general question, but how do you subtract the gravity from the acceleration synchronous?
> 
> Im pulling both once per second but it seems that gravity starts a bit later than Accelerator so when I calculate the linear Acceleration it's slightly off

Hey, nice question unfortunately the site document is not update based document attached to source code so I will refer you to it

Here's the new API gravity:
[API-DOC](https://github.com/react-native-sensors/react-native-sensors/blob/master/docs/API.md#gravity-observablex-number-y-number-z-number-timestamp-string)

And here I have explained exactly how you can exclude it by RxJS
[USAGE-DOC](https://github.com/react-native-sensors/react-native-sensors/blob/master/docs/Usage.md#raw-device-acceleration)

if there is anything else you need, let me know ;)

Tim4497, 2022-03-01:
> And here I have explained exactly how you can exclude it by RxJS
> [USAGE-DOC](https://github.com/react-native-sensors/react-native-sensors/blob/master/docs/Usage.md#raw-device-acceleration)

Nice solution but sadly when one Sensor updates before the other it will generate a new Output, I get two values in the same interval, the solution can be to just take one of the two calculated values per interval, but I guess there is no way to really clock both sensors to the same millisecond?

Ehsan0Hejazi, 2022-03-02:
> Nice solution but sadly when one Sensor updates before the other it will generate a new Output, I get two values in the same interval, the solution can be to just take one of the two calculated [...]

I know what you mean,
to clarify there are three APIs for dealing with acceleration

1-Total Acceleration which is (device acceleration + gravity acceleration )
2-Gravity acceleration standalone
3-linear acceleration which is (Total acceleration - Gravity acceleration)

based on an issue which linear acceleration API is not supported by a few old android devices I decided not to include it into package and instead found a way by RxJS to combine streams for sure you will lose a few milliseconds and it's not a definite solution to problem but based on my  project was practical enough to cover my needs

if you don't mind device support range , let me know to make an update in a few days and add this API as well!

Tim4497, 2022-03-02:
I'm building an app for collecting sensor data from my phone to run them against an AI. 
I don't know if I really need the acceleration in the first place but I want to collect one data point per interval, so its hard to decide which of the two from combine_latest is more relevant.

But overall I guess a lot of people don't mind the few old android devices :) 

It also seems that this package is the most complete and user-friendliest sensor API so I guess it wouldn't be bad to add this feature! 

I would do it myself but I'm sadly not into contributing to open source software, so thanks for taking your time to help me out with this one ❤️

Ehsan0Hejazi, 2022-03-02:
> I'm building an app for collecting sensor data from my phone to run them against an AI. I don't know if I really need the acceleration in the first place but I want to collect one data point per [...]

it seems interesting, 
I will add this feature in a few days and will let you know here
stay tuned ;)

adeshchopade, 2022-04-20:
@DanielMSchmidt @Ehsan0Hejazi 

Thanks for all the work that you have done on this library.

It would be great if you could add linearAcceleration as well.

Right now I'm subtracting gravity values from the raw accelerometer, however, there is a time difference between data returned by both of the sensors, sometimes upto 1 second which is affecting calculations in the app I'm working on.

Ehsan0Hejazi, 2022-04-22:
> calculations

Your welcome,
does this calculations make a meaningful effect on your application logic?!
as I have mentioned before linearAcceleration might not be accessible on some devices but for sure we can add it to the package,  so you need to check to make sure It responses to you with correct data otherwise the only alternative would be the same dummy calculations!

adeshchopade, 2022-04-23:
Yes. We're working on an app that identifies driving behaviour. We use the accelerometer values as one of the parameters.
Having even a 100ms delay in the accelerometer and gravity values is adding a huge margin of error.

Agreed that linearAcceleration might not be available in older / budget android devices, but it does make sense to add it to the library for majority of the devices that do support this feature. 

Hope this is added in a future release.

Ehsan0Hejazi, 2022-04-23:
> Yes. We're working on an app that identifies driving behaviour. We use the accelerometer values as one of the parameters. Having even a 100ms delay in the accelerometer and gravity values is adding [...]

Ok , I will make a pr for upcoming week with the requested feature!

adeshchopade, 2022-04-26:
Thanks. Would be great if you could do that.