react-native-sensors/react-native-sensors #403: Android does not emit data on subscribe

Ehsan0Hejazi, 2021-07-29:
On last version ios is working fine but android does not emit any data

qasimmehdi, 2021-08-01:
I am having the same issue in 7.3. First I installed the package and ran the app without using link and the subscription was not logging any valyes. Then I ran link command and tried again it is still not working. Tried to downgrade to version 7.2 then I got an error Native modules for sensors not available. Did react-native link run successfully?
Using
react-native 0.64.2

Ehsan0Hejazi, 2021-08-02:
I have added a new feature to the package and some references on android does not work correctly I will try to solve the issue
If you don't need the raw values of acceleration feel free to use the old versions

Ehsan0Hejazi, 2021-08-02:
but if you need gravity to exclude it form accelerometer I will give you a temporary solution until the next update,

node_module > react-native-sensors > android > src > main > java > com > sensors > RNSensorsPackage.java

replace 

        -new RNSensor(reactContext, "RNSensorsGyroscope", Sensor.TYPE_GYROSCOPE),
        -new RNSensor(reactContext, "RNSensorsAccelerometer", Sensor.TYPE_ACCELEROMETER),
        -new RNSensor(reactContext, "RNSensorsGravity", Sensor.TYPE_GRAVITY),
        -new RNSensor(reactContext, "RNSensorsMagnetometer", Sensor.TYPE_MAGNETIC_FIELD),
        -new RNSensor(reactContext, "RNSensorsBarometer", Sensor.TYPE_PRESSURE),
        -new RNSensor(reactContext, "RNSensorsOrientation", Sensor.TYPE_ROTATION_VECTOR)

with

        +new RNSensor(reactContext, "Gyroscope", Sensor.TYPE_GYROSCOPE),
        +new RNSensor(reactContext, "Accelerometer", Sensor.TYPE_ACCELEROMETER),
        +new RNSensor(reactContext, "Gravity", Sensor.TYPE_GRAVITY),
        +new RNSensor(reactContext, "Magnetometer", Sensor.TYPE_MAGNETIC_FIELD),
        +new RNSensor(reactContext, "Barometer", Sensor.TYPE_PRESSURE),
        +new RNSensor(reactContext, "Orientation", Sensor.TYPE_ROTATION_VECTOR)

rebuild your app and now it must work as before and you will have raw gravity as well ;)

qasimmehdi, 2021-08-02:
I am using magnetometer in the useEffect of my screen
```js
import { magnetometer } from "react-native-sensors";

const subscription = magnetometer.subscribe(({ x, y, z, timestamp }) =>
  console.log({ x, y, z, timestamp })
);
```

and the above code does not log anything. But I logged subscription variable and it has some data in it so `magnetometer.subscribe` line is executing.

am2619015, 2021-08-05:
Same issue here, gladly I just downgrade version to 7.2.0, and it's working, not version above this one is working properly in android

sxsx2yzyz, 2021-08-06:
It works for me
node_modules/react-native-sensors/src/sensors.js
```
const listenerKeys = new Map([
  ["accelerometer", "Accelerometer"],
  ["gyroscope", "Gyroscope"],
  ["magnetometer", "Magnetometer"],
  ["barometer", "Barometer"],
  ["orientation", "Orientation"],
  ["gravity", "Gravity"],
]);
```
Change to
```
const listenerKeys = new Map([
  ["accelerometer", "RNSensorsAccelerometer"],
  ["gyroscope", "RNSensorsGyroscope"],
  ["magnetometer", "RNSensorsMagnetometer"],
  ["barometer", "RNSensorsBarometer"],
  ["orientation", "RNSensorsOrientation"],
  ["gravity", "RNSensorsGravity"],
]);
```
> but if you need gravity to exclude it form accelerometer I will give you a temporary solution until the next update,
>
> node_module > react-native-sensors > android > src > main > java > com > sensors [...]

DanielMSchmidt, 2021-08-07:
:tada: This issue has been resolved in version 7.3.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-sensors/v/7.3.2)
- [GitHub release](https://github.com/react-native-sensors/react-native-sensors/releases/tag/v7.3.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit d51f758f (2021-08-06): fix: fix listener naming

Closes #403 #404
PR #405 (2021-08-07): fix: fix listener naming
Closes #403 #404
closed 2021-08-07 by DanielMSchmidt