react-native-sensors/react-native-sensors #404: Megnetometer not return data in console

Rafay3198, 2021-08-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

1.  Go to '...'
2.  Click on '....'
3.  Scroll down to '....'
4.  See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**

- React Native Version: [e.g. 55.0.1]
- Device [e.g. real iOS device, android simu]
- Version [e.g. 3.0.0-rc.2]

**Additional context**
Add any other context about the problem here.

Ehsan0Hejazi, 2021-08-04:
node_module > react-native-sensors > android > src > main > java > com > sensors > RNSensorsPackage.java

replace

    -new RNSensor(reactContext, "RNSensorsGyroscope", Sensor.TYPE_GYROSCOPE),
    -new RNSensor(reactContext, "RNSensorsAccelerometer", Sensor.TYPE_ACCELEROMETER),
    -new RNSensor(reactContext, "RNSensorsGravity", Sensor.TYPE_GRAVITY),
    -new RNSensor(reactContext, "RNSensorsMagnetometer", Sensor.TYPE_MAGNETIC_FIELD),
    -new RNSensor(reactContext, "RNSensorsBarometer", Sensor.TYPE_PRESSURE),
    -new RNSensor(reactContext, "RNSensorsOrientation", Sensor.TYPE_ROTATION_VECTOR)
with

    +new RNSensor(reactContext, "Gyroscope", Sensor.TYPE_GYROSCOPE),
    +new RNSensor(reactContext, "Accelerometer", Sensor.TYPE_ACCELEROMETER),
    +new RNSensor(reactContext, "Gravity", Sensor.TYPE_GRAVITY),
    +new RNSensor(reactContext, "Magnetometer", Sensor.TYPE_MAGNETIC_FIELD),
    +new RNSensor(reactContext, "Barometer", Sensor.TYPE_PRESSURE),
    +new RNSensor(reactContext, "Orientation", Sensor.TYPE_ROTATION_VECTOR)
rebuild your app and now it must work as before and you will have raw gravity as well ;)

Ehsan0Hejazi, 2021-08-04:
@DanielMSchmidt please solve this issue, thanks

DanielMSchmidt, 2021-08-08:
This should be fixed by #405 so in version `7.3.2`

closed 2021-08-08 by DanielMSchmidt