react-native-sensors/react-native-sensors #407: `RNSensorsAccelerometer` is not a supported event type for RNSensorsAccelerometer

kyoz, 2021-08-19:
**Describe the bug**

Version 7.3.2 will throw this error when try listeners for sensors:

```
`RNSensorsAccelerometer` is not a supported event type for RNSensorsAccelerometer. Supported events are: `Accelerometer`
```

Maybe duo to the change of listener names in [this commit](https://github.com/react-native-sensors/react-native-sensors/commit/61c1d12f47c51f9e619f89a3299d198b74ca1387)

7.3.1 and below still working

**Expected behavior**

Should working without error

**Desktop (please complete the following information):**

- React Native Version: 0.64.2
- Device: Iphone 5S (64GB)
- Version: 7.3.2

kurovskyi, 2021-08-19:
The same thing on RN 0.64.2

kurovskyi, 2021-08-19:
I've tried go to "node_modules/react-native-sensors/src/sensors.js" and rename "RNSensorAccelerometer" to "Accelerometer" in listenerKeys var:
`["accelerometer", "Accelerometer"],`

Waiting for the fix.

DanielMSchmidt, 2021-08-19:
:tada: This issue has been resolved in version 7.3.3 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-sensors/v/7.3.3)
- [GitHub release](https://github.com/react-native-sensors/react-native-sensors/releases/tag/v7.3.3)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 3c865941 (2021-08-19): fix: adjust ios sensor event name to match android

Closes #407
PR #408 (2021-08-19): fix: adjust ios sensor event name to match android
Closes #407
commit 741e59e0 (2021-08-19): fix: adjust ios sensor event name to match android (#408)

Closes #407
closed 2021-08-19 by DanielMSchmidt