react-native-sensors/react-native-sensors #413: Uncaught Error: Unknown (iOS Simulator)

phen0menon, 2021-11-09:
**Describe the bug**
Getting critical error on iOS simulator when subscribing to accelerometer. The error can be hidden. But it appears every time you start the emulator, which is terribly inconvenient. It also produces an error in the console:

```
 ERROR  Sensor accelerometer is not available
```

[image]

**To Reproduce**
Steps to reproduce the behavior:

1. Install latest version of react-native-sensors (7.3.3)
2. Run iOS app on simulator (tested on 15.0)

**Expected behavior**
Display the simulator error only in the console 

**Desktop (please complete the following information):**

- React Native Version: 0.64.1
- Device iOS Simulator (iOS 15.0)
- Version 7.3.3

grgmgd, 2021-11-18:
I solved mine by catching the error using RxJs

DenysLopatynskyiB, 2022-06-13:
+1 to this.
Once you call accelerometer.subscribe you get the error

[image]