react-native-sensors/react-native-sensors #414: Performance Issues on 10ms updates

alissayarmantho, 2021-12-09:
Hello, I am trying to use the accelerometer and gyroscope sensors to do fall detection. However, this requires updates every 10ms or at most 100ms which I believe saturates the react-native bridge. Is there a possible solution for this? Thanks for the help!

DanielMSchmidt, 2021-12-10:
Hey, I think you'd need to have the business logic in the native land, so in Android or iOS directly. I'm not aware on ways for libraries to  optimize the bridge usage. 

One thing we could try is to add a level of filter to the API, but we'd need to be able to express it statically, e.g. new Gyroscope({minThreashold: 42}) and it only fires if that threashold is met and therefore leaves the bridge alone for most of it's lifecycle. Not sure if that's something you want to do though 🤔

alissayarmantho, 2021-12-10:
I see, thank you for the suggestion!

jamonholmgren, 2022-04-26:
With the React Native new architecture & JSI, this problem could be worked around. RN 0.68 has support (behind a flag) and RN 0.69 will have React 18.

jainharsh21, 2022-07-29:
@alissayarmantho could you please assist me in knowing how exactly are you using this package for fall detection, our team needs to integrate in an app as well, it would be very much appreciated

alissayarmantho, 2022-07-29:
I use the gyroscope and acceleration to calculate a value. This value is used to determine if there's a fall detection or not.

The calculation formula is from some research paper.

jainharsh21, 2022-07-29:
@alissayarmantho thank you for the reply, is it possible for you to share the formula or the code snippet, it'll be really helpful, my use case is to report if a fall was detected or not, thanks in advance.

alissayarmantho, 2022-07-29:
Ah, I think you can google them somewhere online. I've forgotten where it came from and the accuracy was not that ideal either. The code was quite long too so I don't think I can share that here.

jainharsh21, 2022-07-29:
Okay thank you