react-native-sensors/react-native-sensors #416: Error with addListener and removeListeners on Android

AidenForrest, 2022-01-28:
Hello,

I have recently installed this module on my React Native project and it works perfectly fine on iOS, but on Android I get a never-ending stream of the following two warning messages which freezes the app completely:

```
WARN  `new NativeEventEmitter()` was called with a non-null argument without the required `removeListeners` method.
WARN  `new NativeEventEmitter()` was called with a non-null argument without the required `addListener` method.
```

Any help would be greatly appreciated.

RN version: 0.67.1
react-native-sensors version: 7.3.3
Android version: 9

AidenForrest, 2022-01-28:
Fixed, was incorrectly installed

wibimaster, 2022-02-27:
@AidenForrest Can I know how you fix it ? I have the same error after a basic install :
```sh
npm install react-native-sensors --save
react-native link react-native-sensors
```

AidenForrest, 2022-02-27:
@wibimaster I don’t think I ever managed to completely fix it but a work around was disabling the errors by either ignoring them by using:

`import { LogBox } from 'react-native';
LogBox.ignoreLogs(['Warning: ...']); // Ignore log notification by message `

Or another method that prevented the freezing was running the Android app as release rather than debug. 

Hope one of these workarounds works for you.

sedot42, 2022-03-23:
This is helpful: https://stackoverflow.com/questions/69538962/new-nativeeventemitter-was-called-with-a-non-null-argument-without-the-requir

A pull request like this might fix this: https://github.com/microsoft/appcenter-sdk-react-native/pull/942/files

Karemhz, 2022-03-24:
Same problem

am2619015, 2022-10-10:
Instead of ignoring all warnings, probably the best approach will be add on index.js
import { LogBox } from 'react-native';
LogBox.ignoreLogs(['new NativeEventEmitter']); 
However, it is not a fix, the future library should support these warnings, this happens in new versions of RN like 0.70.2

powerpivar, 2022-11-08:
Same problem for me, any updates?

`"react-native-sensors": "7.3.5",
"react-native": "0.67.4",
`

closed 2023-07-02 by AidenForrest