react-native-sensors/react-native-sensors #417: Running with hermes engine causes app to crash

rubina-shakhkyan, 2022-02-04:
**Describe the bug**
The crash can be fixed by removing `toLocaleLowercase` function. 

**To Reproduce**
Steps to reproduce the behavior:

Integrate the package in an app, where `react-native 0.66` and `hermes 0.9.0` are used.

**Expected behavior**
Should not crash
**Desktop (please complete the following information):**

- React Native Version: [0.66.0]
- Device [any android device]
- Version [7.3.3]

DanielMSchmidt, 2022-02-04:
Ah, didn't know they are not supporting that :) I'm more than happy to review a Pr 👍

rubina-shakhkyan, 2022-02-04:
> Ah, didn't know they are not supporting that :) I'm more than happy to review a Pr 👍

Here you go 🙂 https://github.com/react-native-sensors/react-native-sensors/pull/418

I just quickly removed lowercase-ing.

closed 2022-02-10 by rubina-shakhkyan