react-native-sensors/react-native-sensors #426: Error: Native modules for sensors not available. Did react-native link run successfully

carltian, 2022-03-07:
**Describe the bug**
when i use it on ReactNative, it can't work, and report Error: Native modules for sensors not available. Did react-native link run successfully

**To Reproduce**

1.  yarn add react-native-sensors
2.  react-native link react-native-sensors
3.  react-native run-android
4.  See error: Native modules for sensors not available. Did react-native link run successfully

- React Native Version: 0.67.3
- Device  android simu
- Version 7.3.5

MordiukRoman, 2022-11-16:
have the same issue on android emulator, react native 0.67.4

littlecircle0730, 2022-11-21:
Did this be fixed?

Koxx3, 2023-06-15:
I have the same issue when using react-native-sensor-fusion

sayan-avantai, 2025-04-09:
Any update? Still I am having this issue.