react-native-sensors/react-native-sensors #435: Ios error on installation

VanLMC, 2022-07-20:
First i was running the app on ios and getting the same error as in this issue https://github.com/react-native-sensors/react-native-sensors/issues/413. Then i thought it could possibly be because i missed some step in the ios installation guide. But after following the steps in the guide here : https://react-native-sensors.github.io/docs/Installation.html the app doesn't even build anymore. I get the following error on xcode:

[image]

Am i doing something wrong on the installation? Did it change? Or is this a bug?

    "react-native": "0.66.3",
    "react-native-sensors": "^7.3.3",

rowannadon, 2022-07-23:
I have the same issue, except the missing file is React/RCTBridge.h

mertbo, 2023-02-01:
In my case, I accidentally applied both manual and automatic installations. I didn't get this error when I just did the automatic installation: https://react-native-sensors.github.io/docs/Installation.html#automatic-installation