react-native-sensors/react-native-sensors #448: Crash when reloading

mo22, 2023-06-04:
Getting the following error during reload:
```
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Error when sending event: RNSensorsBarometer with body: {
    pressure = "1013.959655761719";
    timestamp = 1685887033002;
}. RCTCallableJSModules is not set. This is probably because you've explicitly synthesized the RCTCallableJSModules in RNSensorsBarometer, even though it's inherited from RCTEventEmitter.'
```

proposed solution: add to all modules:

```
- (void)stopObserving {
    [self stopUpdates];
}
```

skizzo, 2023-10-29:
Were you able to fix this? Thanks

gregbrinker, 2023-10-31:
This fix from @zhongwuzw worked for me. You can implement the same pattern in whichever sensors you're using

https://github.com/react-native-sensors/react-native-sensors/pull/458

zhongwuzw, 2023-11-01:
I would update the PR to fix this too.