react-native-sensors/react-native-sensors #452: "Sensor orientation is not available" for some Android devices

ttrvdw, 2023-08-10:
**Describe the bug**
Orientation is not available for devices that do not have some sensors. e.g. a device with only accelerometer will return "Sensor orientation is not available" error when subscribing to "orientation". 

Tested setup:
Samsung Galaxy A04s with accelerometer but no gyroscope or magnetometer. 
Not happening on Samsung Galaxy S8 and iPhone XR.

**To Reproduce**
Steps to reproduce the behavior:

Take phone with accelerometer but no gyroscope nor magnetometer (e.g. Samsung A20).

```
import {
  orientation,
  SensorTypes,
  setUpdateIntervalForType,
} from 'react-native-sensors';

const subscription = orientation.subscribe(data => {...})
```

This will return 
`Sensor orientation is not available`

**Expected behavior**
The package should check what sensors are available to make the orientation measurements. I suspect that some sensor fusion is happening under the surface, but the module should check internally what sensors are available for the fusion. In case of single sensor, an orientation estimate, be it noisy, should still be returned. It's up to the end user to decide how to deal with noisy sensor data.

**Desktop (please complete the following information):**

- React Native Version: 0.71.7
- Device: Samsung Galaxy A04s, physical device
- Version: Android 13

ernesdev, 2023-09-28:
I had to solve it in the native Android module, that is the patch that worked for me using the accelerometer sensor.
@ttrvdw 
[react-native-sensors+7.3.6.patch](https://github.com/react-native-sensors/react-native-sensors/files/12751884/react-native-sensors%2B7.3.6.patch)

> **Describe the bug** Orientation is not available for devices that do not have some sensors. e.g. a device with only accelerometer will return "Sensor orientation is not available" error when subscribing to "orientation".
> 
> Tested setup: Samsung Galaxy A04s with accelerometer but no gyroscope or magnetometer. Working without error on Samsung Galaxy S8 and iPhone XR.
> 
> **To Reproduce** Steps to reproduce the behavior:
> 
> Take phone with accelerometer but no gyroscope nor magnetometer (e.g. Samsung A20).
> 
> ```
> import {
>   orientation,
>   SensorTypes,
>   setUpdateIntervalForType,
> } from 'react-native-sensors';
> 
> const subscription = orientation.subscribe(data => {...})
> ```
> 
> This will return `Sensor orientation is not available`
> 
> **Expected behavior** The package should check what sensors are available to make the orientation measurements. I suspect that some sensor fusion is happening under the surface, but the module should check internally what sensors are available for the fusion. In case of single sensor, an orientation estimate, be it noisy, should still be returned. It's up to the end user to decide how to deal with noisy sensor data.
> 
> **Desktop**
> 
> * React Native Version: 0.71.7
> * Device: Samsung Galaxy A04s, physical device
> * Version: Android 13

carltian, 2023-11-17:
> I had to solve it in the native Android module, that is the patch that worked for me using the accelerometer sensor. @ttrvdw [...]

hi, I use your document in my project,  but it report a error
`TypeError: Cannot read property 'DocumentDir' of null`
i find some absolute path in the document, i try to fix it,but failed
[image]

drewdev02, 2024-05-25:
> I had to solve it in the native Android module, that is the patch that worked for me using the accelerometer sensor. @ttrvdw [...]

how install react-native-sensors+7.3.6.patch?

konsnos, 2025-03-06:
I have the same issue. The app crashes. Even adding an error handler doesn't help.

```
const subscription = sensorObservable.subscribe({
        next: data => {
        },
        error: () => console.warn(`Sensor ${type} isn't available`),
      });
```