react-native-sensors/react-native-sensors #456: Build Failed on IOS RN 0.72 RNSensors v7.3.6 - RCTBridge & RCTBridgeModule not found

flikQ, 2023-10-23:
Installation of React-Native-Sensors fails on RN 0.72..

Errors as follows:

- /.../react-native-sensors/ios/RNSensorsOrientation.m:5:9: 'React/RCTBridge.h' file not found
- /.../react-native-sensors/ios/RNSensorsMagnetometer.m:5:9: 'React/RCTBridge.h' file not found
- /.../react-native-sensors/ios/RNSensorsGyroscope.h:3:9: 'React/RCTBridgeModule.h' file not found

**To Reproduce**
Try and install on RN 0.72.5 using YARN & pod install.

Attempted manual and automatic installation independently so they don't conflict.

**Expected behavior**
Install correctly and work.

**Desktop (please complete the following information):**

- React Native Version: 0.72.5
- Device: iOS on Sim and/or Device
- Version: v7.3.6

WookyungYun, 2024-04-18:
same here! did you solve it?

lehung97abc, 2024-05-04:
same here! you can fix it?

WookyungYun, 2024-05-04:
> same here! you can fix it?

I fixed it! In my case, npm install process was the problem.

you can install with '--save'

lehung97abc, 2024-05-04:
> > same here! you can fix it?
> 
> I fixed it! In my case, npm install process was the problem.
> 
> you can install with '--save'

i follow https://react-native-sensors.github.io/docs/Installation.html . but errors as follows:
.../node_modules/react-native-sensors/ios/RNSensorsMagnetometer.m:5:9 'React/RCTBridge.h' file not found

use yarn or nvm --save is same