react-native-sensors/react-native-sensors #461: Accelerometer work good on android, NOT working on iOS

nppull, 2024-05-08:
**Describe the bug**
Accelerometer work good on android, **NOT working on iOS**.

```
const subscription = accelerometer
  .pipe(map(({ x, y, z }) => x + y + z), filter(speed => speed > 20))
  .subscribe(
    speed => console.log(`You moved your phone with ${speed}`),
    error => {
      console.log("The sensor is not available");
    }
  );
```

**Desktop (please complete the following information):**
- React Native: 0.73.5
- Device: iphone xs max, ios 17.4.1
- react-native-sensors: 7.3.6

Anyone have workaround ?

jcndsc, 2024-09-02:
Having the same issue here

konsnos, 2024-12-13:
Try this one from the [documentation](https://react-native-sensors.github.io/docs/API.html)

```
import { accelerometer } from "react-native-sensors";

const subscription = accelerometer.subscribe(({ x, y, z, timestamp }) =>
  console.log({ x, y, z, timestamp })
);
```